foxglove/studio #3375: Error: new playerState was emitted before last playerState was rendered

foxymiles, 2022-05-16:
**Description**
This error occurs sometimes when changing layouts or data sources.

```
Uncaught (in promise) Error: New playerState was emitted before last playerState was rendered.
    at listener (index.tsx?2844:200:1)
    at UserNodePlayer.eval [as _listener] (OrderedStampPlayer.ts?96f9:97:1)
    at UserNodePlayer._emitState (index.ts?3d38:541:1)
    at UserNodePlayer._onPlayerState (index.ts?3d38:523:1)
    at async IterablePlayer.eval [as _listener] (index.ts?3d38:549:1)
    at async IterablePlayer._emitState (IterablePlayer.ts?0830:469:1)
```

- Version: 1.11
- OS: macOS

**Steps To Reproduce**
Rapidly switch among different layouts.

**Expected Behavior**
No error should arise.

**Actual Behavior**
This exception is thrown.

foxymiles, 2022-05-31:
@jtbandes Is this condition severe enough we need to throw an error? Maybe logging it would be sufficient?

https://github.com/foxglove/studio/blob/bef9bdccb6d8d877dda0cda652ae30d8199880f4/packages/studio-base/src/components/MessagePipeline/index.tsx#L313-L315

jtbandes, 2022-05-31:
This indicates a breakage of the Player API contract. I can't really predict what may go wrong if it is violated. Certainly a violation only when switching data sources / layouts is not as bad as a violation during regular playback, so there might not be much harm done right now, but it still seems preferable to fix rather than downgrade to a warning and potentially invite future harm...

PR #3512 (2022-06-09): Prevent concurent state emit in iterable player when setting playback

**User-Facing Changes**
When switching layouts using any of the experimental players the app no longer crashes with an error about emitting playerState before last playerState was rendered.

**Description**
When setting the playback speed for the player the player needs to emit state so downstream components can update with the new playback speed. The iterable player emits the state only when idle, however even when idle the player may be loading blocks and emitting state which would cause concurrent state emits which is in [... cut]
commit b839859f (2022-06-09): Prevent concurent state emit in iterable player when setting playback (#3512)

When setting the playback speed for the player the player needs to emit state so downstream components can update with the new playback speed. The iterable player emits the state only when idle, however even when idle the player may be loading blocks and emitting state which would cause concurrent state emits which is invalid in our pipeline.

This change fixes this behavior by triggering a state change back to "idle" if the player gets a speed update when already in idle. The state change will trigger an emit of th [... cut]
closed 2022-06-09 by defunctzombie