foxglove/studio #3401: Extensions should be able to load static resources

rgov, 2022-05-18:
As far as I can tell, there is no recommended way to bundle a static resource with an extension and reference it at runtime.

I *can* configure Webpack thusly to enable [Asset Modules](https://webpack.js.org/guides/asset-management/) for things like images.

```javascript
module.exports = {
    webpack: (config) => {
        config.module.rules.push({
            test: /\.(png|svg|jpg|jpeg|gif)$/i,
            type: "asset/inline",
        });
        return config;
    },
};
```

The use of `asset/inline` here loads the whole image as a Base64-encoded `data:` URL that gets embedded into the DOM when I write,

```jsx
import MyImage from "./images/foobar.png";
...
<img src={MyImage} />
```

I don't think it's possible to use `asset/resource` which instead inserts the path of the file, like `<img src="./images/foobar.png">`, because the base URI to the extension is not known at runtime.

Is it maybe possible to have the extension loader prepend something like `let __webpack_public_path__ = "...";` to the extension source code before executing it?

rgov, 2022-05-19:
I spent an obscene amount of time toying with this and the solution described above does kind of work. See here: https://github.com/foxglove/studio/compare/main...rgov:rzg/extension-path

In my extension's Webpack `config.ts` I specify `config.output.publicPath = ""` and then before I import anything in my main file, I use

```javascript
declare var __webpack_public_path__: string; 
__webpack_public_path__ = (window as any).foobar;
```

(Why `window.foobar`? I had to figure out some way to smuggle the path into the scope that overrides `__webpack_public_path__`. I'm sure there is a more fitting way to do this.)

rgov, 2022-05-19:
A simpler way to do it, once the extension directory is plumbed through, is to set `config.output.publicPath = "@FOXGLOVE_EXTENSION_PATH@"` and have the ExtensionRegistryProvider use

```javascript
const unwrappedExtensionSource = (await extensionLoader.loadExtension(extension.id))
          .replace("@FOXGLOVE_EXTENSION_PATH@", `file://${extension.directory}/dist/`);
```

Additionally the Content-Security-Policy (in `desktop/main/index.ts`) may need to be tweaked or disabled to allow loading files from the extensions directory.

amacneil, 2022-08-05:
From live discussion:

Problem statement
- monaco is hard to import in webpack (or at least we have no example of this working). [here](https://github.com/foxglove/studio/blob/b9793b1a76daf265ca190a93c1726b8a2bd7808e/packages/studio-base/webpack.ts#L263-L270) is how we do it in studio.
- importing theora ( https://github.com/brion/ogv.js ) with wasm binary is not working because it appears to expect the wasm blob to be available on a URL.

We will create tickets to attempt examples for both of these.