foxglove/studio #3412: cant connect to ROS stream via docker

an99990, 2022-05-19:
**Description**
I am running a docker container in which theres is a ros2 node running. I would like to visualize the images in foxgloves.
I expose the IP adress when running the docker.
`
docker run --expose 8080 -it -p 8080:8080 --runtime nvidia --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix buffer`

OPen connection steps

Foxglove does not seem to be able to connect to the server

[image]

rgov, 2022-05-19:
I think you have two problems:

1. You should be connecting to the Websocket of the Rosbridge server. Port 8080 is set up to be Foxglove's web server port, so Rosbridge must be running on another port. It is usually on port 9090.

2. When you're running in a container, the hostname `localhost` refers to *the container itself* and not to the host operating system. So if you connect to `localhost` it would try to connect to some other service running inside the same container.

    On the host, you can use `ip addr` to find the address of the Docker interface (called something like `docker0`). It is often `172.17.0.1`. Then you could connect Foxglove Studio to `ws://172.17.0.1:8080`.

If you don't need to be using Docker I would suggest using the standalone app.

an99990, 2022-05-19:
hi @rgov thank you for answering so quickly. I have unfortunalty no choice but to run the docker to access my ros node. What do you suggest me to change.
`docker run --expose 9090 -it -p 9090:9090 --runtime nvidia --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix buffer`

?

an99990, 2022-05-19:
should i run the docker differently ? 

[image]

[image]

rgov, 2022-05-19:
Sorry I misunderstood, I thought you were running Foxglove Studio itself in a container.

So your ROS nodes are running inside the container, and this includes a [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) Rosbridge server. Then yes, use `-p 9090:9090`, I don't think you also need `--expose`. 

The port number 9090 is defined here:

https://github.com/RobotWebTools/rosbridge_suite/blob/d9c4f3973da27920fb9963f1717a362b2ba6e7e1/rosbridge_server/scripts/rosbridge_websocket.py#L110

Then `ws://localhost:9090` should work from Foxglove Studio.

an99990, 2022-05-20:
HI thank you @rgov for your answer
I did as you told and it does not seem to be working .

just so were on the same page, the docker that i run has the ros2 node running

how i run it
`docker run -it -p 9090:9090 --runtime nvidia --privileged --gpus all -it --ipc=host --ulimit memlock=-1 --ulimit stack=67108864 zed_torch`

It keeps waiting for connection 

[image]

rgov, 2022-05-20:
And the container and Foxglove Studio are running on the same host system? Can you provide proof that the Rosbridge server is actually running inside the container (e.g., `docker exec <your running container id> ps -awx`, `docker exec <your running container id> netstat -plnt`)

Are you using the Foxglove Studio app or using it from the web browser? Open the Developer Tools (in app, View -> Advanced -> Toggle Developer Tools) and look at the Console. Does it give any more information about not being able to connect?

rgov, 2022-05-24:
There are no ROS nodes running inside your `zed_torch` container.

an99990, 2022-05-24:
[image]

rgov, 2022-05-24:
That looks slightly better, but as mentioned you need to also run a rosbridge_server node from the [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) package inside the container.

an99990, 2022-05-24:
[image]

rgov, 2022-05-24:
You didn't pass `-p 9090:9090` to the container that is running your rosbridge server. Also it is recommended that you run your nodes and the rosbridge server in the same container.

If you need further support, I suggest closing this issue and asking on https://answers.ros.org/. The issue is not with Foxglove Studio, but seem to be misconceptions about how Docker and/or ROS work.

closed 2022-05-24 by an99990