foxglove/studio #3417: The foxglove studio Log panel does not show log message rightly.

xyzxyzxyzxyzxz, 2022-05-20:
**Description**
When I use ros2 to publish log message,the foxglove studio log panel can not show the log level rightly and the timestamp is always "8:00:00.000 AM CST". And if I use the filter such as >=ERROR,the foxglove studio show all the message no matter it is WARN or INFO.
[image]
And this is what is shown in the terminal.
[image]

I think it is the problem of foxglove studio.Because if I use turtlesim,the problem appears similarly.
[image]

Then I tried to publish the message rcl_interfaces/msg/Log manually,and set the level which  equals 3,the foxglove studio can filtrate the message,but the foxglove studio still show the message wrongly.
[image]

The problem only appears in ROS2.If I use turtlesim in ROS1 ,the foxglove studio show the log message rightly.

- Version:1.10.1
- OS:Ubuntu 18.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):ROS 2 native
- ROS distro (if applicable):dashing

**Steps To Reproduce**

The following is my code
node_a.h
```c++
#pragma once

#include <chrono>

#include <rclcpp/rclcpp.hpp>

#include "task_interfaces/msg/adc_trajectory.hpp"
#include "task_interfaces/msg/planning_analysis.hpp"

#include "rcl_interfaces/msg/log.hpp"

#include "rclcpp/logger.hpp"

namespace ros2 {
using task_interfaces::msg::PlanningAnalysis;
using task_interfaces::msg::ADCTrajectory;
class NodeA : public rclcpp::Node{
 public:
  NodeA(const std::string & name);
  ~NodeA() = default;
  double count_=0;
  rclcpp::Logger mylog=this->get_logger();

 protected:
  rclcpp::Subscription<PlanningAnalysis>::SharedPtr planning_analysis_sub_;
  rclcpp::Publisher<rcl_interfaces::msg::Log>::SharedPtr adc_trajectory_pub_;
  rclcpp::TimerBase::SharedPtr timer_;
 protected:
 void OnPlanningAnalysis(const PlanningAnalysis::SharedPtr msg);
 void TimerCallback();
};

}  // namespace ros2
```
node_a.cpp
```c++
#include "node_a.h"

namespace ros2 {
NodeA::NodeA(const std::string& name)
    : Node{name} {
  using rclcpp::QoS;
  using SubAllocT =
      rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void>>;
  using PubAllocT = rclcpp::PublisherOptionsWithAllocator<std::allocator<void>>;

  planning_analysis_sub_ = create_subscription<PlanningAnalysis>(
      "PlanningAnalysisTest", QoS{10},
      [this](const PlanningAnalysis::SharedPtr msg) {
        OnPlanningAnalysis(msg);
      },
      SubAllocT{});
  adc_trajectory_pub_ =
      create_publisher<rcl_interfaces::msg::Log>("mylog", QoS{10}, PubAllocT{});
  timer_ = create_wall_timer(std::chrono::milliseconds(50), std::bind(&NodeA::TimerCallback, this));
}

  void NodeA::OnPlanningAnalysis(const PlanningAnalysis::SharedPtr msg){
    RCLCPP_INFO(this->get_logger(), "I heard: 'frame_update_time:%f'",msg->frame_update_time);
  }

  void NodeA::TimerCallback(){
    //while(rclcpp::ok())
    {
      /*getchar();
      RCLCPP_DEBUG(mylog,"this is a DEBUG");
      getchar();
      RCLCPP_INFO(mylog,"this is a INFO");
      getchar();
      RCLCPP_WARN(mylog,"this is a WARN");
      getchar();
      RCLCPP_ERROR(mylog,"this is a ERROR");
      getchar();
      RCLCPP_FATAL(mylog,"this is a FATAL");*/

      rcl_interfaces::msg::Log logmsg;
      logmsg.level=3;
      uint64_t sys_time=std::chrono::duration_cast<std::chrono::microseconds>(std::chrono::system_clock::now().time_since_epoch()).count();
      logmsg.stamp.sec=sys_time/1000000;
      logmsg.stamp.nanosec=sys_time%1000000*1000;
      logmsg.name="NodeA";
      logmsg.msg="this is a warning";
      adc_trajectory_pub_->publish(logmsg);
    }

  }
  }  // namespace ros2
```
maina.cpp
```c++

#include <rclcpp/rclcpp.hpp>

#include "node_a.h"
using ros2::NodeA;
int main(int argc, char* argv[]) {
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<NodeA>("NodeA"));
  rclcpp::shutdown();
  return 0;
}
```
When I use RCLCPP_DEBUG to publish the message,I change `adc_trajectory_pub_ =
      create_publisher<rcl_interfaces::msg::Log>("mylog", QoS{10}, PubAllocT{});` in node_a.cpp  to `adc_trajectory_pub_ =
      create_publisher<ADCTrajectory>("ADCTrajectoryTest", QoS{10}, PubAllocT{});` and comment some code ,and I also change the file node_a.h similarly.

jtbandes, 2022-05-20:
Thanks for the report. Can you record an example bag file that demonstrates the problem?

xyzxyzxyzxyzxz, 2022-05-23:
I record two ros2 bag.
In the first one, I manually publish the log message of type `rcl_interfaces::msg::Log` on topic `/mylog`.
[manually_publish_bag.tar.gz](https://github.com/foxglove/studio/files/8750762/manually_publish_bag.tar.gz)
In the second one,I use `RCLCPP_INFO()`,`RCLCPP_WARN` and so on.

[RCLCPP_INFO_bag.tar.gz](https://github.com/foxglove/studio/files/8750778/RCLCPP_INFO_bag.tar.gz)
@jtbandes

jtbandes, 2022-05-23:
Thanks for the bag files. Fixed in https://github.com/foxglove/studio/pull/3427

closed 2022-05-23 by jtbandes

xyzxyzxyzxyzxz, 2022-05-25:
Thanks for fixing the problem.The Log message can be shown rightly, but the filter still can not be used.
[image]
.