foxglove/studio #3437: Plot Panel Y-limit bugs

jackiescanlon, 2022-05-24:
**Description**
There are a few bugs in the Y-limits settings for Plot panels. I grouped them altogether in this issue but am happy to separate them if that would be better.

- Version: 1.12.0
- OS: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Seems to be data source agnostic - I am able to replicate the issues even when not connected to a data source. For my example below I used a ROS 1 live connection.
- ROS distro (if applicable): Noetic

**Steps To Reproduce and Expected Behavior**

1. Create a new layout and add a Plot panel
2. Plot something
3. Open the panel settings and look at Y-min and Y-max.

**Auto-scale**

**Expected result:** I would've expected the defaults to be `auto` and `auto`, with the plot auto-scaling.
**Actual result:** The plot is auto-scaling as expected, but the Y-min and Y-max are reading back `0` and `0` instead of `auto` and `auto`. This is confusing, especially because if you backspace the `0`s they will turn to `auto`. 
**Workaround:** The only way to set an actual limit of 0 is to backspace the 0 and retype it in.

[image]

[image]

**Negative values**

4. Now, in the Y-min box, backspace the 0 and attempt to type -10.

**Expected result:** As I type `-10`, it appears in the textbox, and the minimum on the plot goes to `-10`.
**Actual result:** A `-` is not allowed in the textbox when the textbox is empty. 
**Workaround:** Type `10`, and then move the cursor to the front to type the `-`. It seems to accept the `-` as long as the textbox already has other text in it and isn't empty.

**Reversed values**

5. Attempt to set the Y-min to 10 and the Y-max to -10 (using the above workaround to type the `-`). I've accidentally done this many times, reversing the Y-min and the Y-max (I think my brain wants to have the Y-max on the top because its setting the value for the top of the graph).

**Expected result:** Studio throws an error at me, because my Y-max is less than my Y-min. Or even better, Studio figures out that i just reversed them and plots [10, 10].
**Actual result:** Foxglove does figure out that I reversed them and plots [-10, 10], which is really helpful. However, for some reason it removes all the horizontal lines on my plot :/
**Workaround:** Don't mix up the Y-min and Y-max :) Now that I know what is going on its easy to workaround, but for awhile I wasn't sure why all my horizontal lines were going away.

[image]

foxymiles, 2022-05-24:
Thanks very much for the detailed bug report!

I've merged a patch that I think should fix this. Let me know if you're still having issues.

jackiescanlon, 2022-05-24:
Wow, that was fast!! I will look for it in a future release. Thank you!

jackiescanlon, 2022-05-26:
I just checked out v1.12.1 and it looks like my first two issues with autoscaling and negative numbers is fixed. Thanks so much!

The third issue, where the horizontal lines disappear, is still happening though.

foxymiles, 2022-05-31:
Thanks for verifying the fix. I'm a little hesitant to silently swap the values of ymin & ymax if the user enters ymax <= ymin, especially since the logic for handling graph bounds is already fairly complex and involves RPC calls to a web worker.

Maybe it would be sufficient to clearly flag the input as an error if ymax <= ymin?

jackiescanlon, 2022-06-06:
Yep, that makes sense to me. Just tested the latest merge, looks good. Thanks!