foxglove/studio #3456: The Log message can be shown rightly, but the filter still can not be used.

263868161, 2022-05-26:
Thanks for fixing the problem.The Log message can be shown rightly, but the filter still can not be used.

[image]

_Originally posted by @xyzxyzxyzxyzxz in https://github.com/foxglove/studio/issues/3417#issuecomment-1136612711_

xyzxyzxyzxyzxz, 2022-05-30:
In ROS1, I run the turtlesim. If I choose ">=INFO" ,the log panel will show the message including WARN and INFO and so on.
[image]
If I choose ">=WARN", the INFO message will not be shown.
[image]
Though I choose ">=ERROR", the WARN message is still here.
[image]
But if I choose ">=FATAL", there will be nothing.
[image]

In ROS2, I use the sample bag. No matter what I choose, it will show all the message.
[image]
[image]

I don't know if the method I used is wrong.

foxymiles, 2022-05-31:
It looks like the ROS2 log levels are defined differently from the ROS1 log levels:

https://github.com/ros2/rcutils/blob/1570091b559fe605fee5cfdf4a6a97f412836656/include/rcutils/logging.h#L167-L175

https://github.com/foxglove/studio/blob/bef9bdccb6d8d877dda0cda652ae30d8199880f4/packages/studio-base/src/panels/Log/types.ts#L9-L16

I guess we'll have to some some player-level awareness of what the numeric log levels should be?

PR #3468 (2022-06-01): Map log level severity values by message datatype
**User-Facing Changes**
None

**Description**
This allows the log panel message filtering logic to handle datasources that use different numeric values to indicate different message severity levels. 

This implementation uses the message datatype to determine if log level mapping is necessary.

Fixes #3456