foxglove/studio #3459: Auto-complete oddities when filtering fields

jameskuszmaul-brt, 2022-05-26:
**Description**

Under some circumstances, attempting to use the auto-completion on vector fields (as in adding lines for plots or trying to show a particular field in the Raw Messages panel) works poorly.

The below is all based on the state of studio.foxglove.dev as of 2022.05.26 (1.12.1-dev), using 
[log.mcap.gz](https://github.com/foxglove/studio/files/8780469/log.mcap.gz)

**Steps To Reproduce**
1. Fill in `"/localizer frc971.IMUValuesBatch".readings[:]` -- no auto-complete suggestions will be forthcoming
2. Start typing `imuvalues`. The autocomplete suggests will include `"/localizer frc971.IMUValuesBatch".readings.accelerometer_x.` (which is incorrect in two ways--(1) that `readings` is a vector and (2) that `accelerometer_x` is a scalar field and so the trailing `.` is misleading).

**Expected Behavior**
1. Once you fill in `[:]` autocompletion should still work.
2. Autocompletion options should always be valid.

**Actual Behavior**

jameskuszmaul-brt, 2022-05-26:
Oh and there appears to be a max limit on the number of allowed auto-complete suggestions. E.g., if you fill in `"/imu/aos aos.timing.Report".` it will not show any options corresponding to the `watchers` vector (this makes breadth-first navigation of the auto-complete suggestions hard, but is less of a bug than the original issues I listed).

foxymiles, 2022-06-06:
Yeah making your id an int will at least give you some suggestions. It looks like offering suggestions following a user-entered `[:]` is going to require reworking the autocomplete logic a bit given our current grammar.

defunctzombie, 2022-06-13:
The oddities described here have been fixed in #3500 so I'm closing this bug.

We provide no completion for slices currently so that part would be a feature request.

closed 2022-06-13 by defunctzombie