foxglove/studio #3463: Watching allFrames crashes Foxglove

f-o-w-l, 2022-05-27:
Whenever I call `context.watch("allFrames")` then subscribe to a topic with over 1GB of messages, `allFrames` is parsed until Foxglove crashes.

- Version: 1.12.1
- OS: PopOS 20.04 LTS 64-bit
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag file
- ROS distro (if applicable): N/A

**Steps To Reproduce**
Call `context.watch("allFrames")` then subscribe to a topic with over 1GB of messages. Foxglove Studio crashes.

Another related issue is that `RenderState` properties cannot be _un_-watched. Steps to reproduce:

1. Add a log message like `console.log("Panel rendered")` in the body of the entry panel component receiving the `PanelExtensionContext` (logging on every re-render)
2. Conditionally call `context.watch("allFrames")` in the `useLayoutEffect`, eg. `if (someBoolean) context.watch("allFrames")` and adding `someBoolean` to the `useLayoutEffect` dependency list
3. Change `someBoolean` to false from somewhere in the extension
4. Subscribe to a new topic/topics with over 1GB of messages
5. Observe that "Panel rendered" will be logged over and over until Foxglove crashes, even though `context.watch("allFrames")` was not called in the most recent `useLayoutEffect` call. Even though `allFrames` is not included in the `RenderState` -- `renderState.allFrames` is correctly undefined after the new subscription(s) -- it seems to be getting populated in the background anyway.

**Expected Behavior**
- Parsing large topics for `allFrames` should not crash Foxglove Studio
- If `context.watch("allFrames")` is not called in the latest render of the `useLayoutEffect`, `RenderState` properties should be unwatched, or they should be able to be unwatched manually.

**Actual Behavior**
Described above.

foxymiles, 2022-06-01:
@jtbandes @jhurliman @defunctzombie Looks like maybe we need an `unwatch` call in our extension API? 

Or we need to change the semantics of `watch` to replace the set of watched properties instead of being purely additive:

https://github.com/foxglove/studio/blob/c5a8364ff1321b63a72847947d7e4e0ed8c86a12/packages/studio-base/src/components/PanelExtensionAdapter.tsx#L413-L415

f-o-w-l, 2022-06-02:
Even though I feel like fixing Foxglove crashing would be the ideal solution, that's probably not an easy task, so yeah, for now it would at least be easy to add a sanity check like `if (renderState.allFrames.length > 5000)` in `context.onRender` that would call `context.unwatch` and display an error message.

foxymiles, 2022-06-03:
@defunctzombie What do you think about extending the semantics of `watch` so that if you pass an array that completely replaces the existing set of watched fields? Passing a single value would be continue to be purely additive as it is now. This would make it work more like `subscribe` currently does.

defunctzombie, 2022-06-08:
> @defunctzombie What do you think about extending the semantics of `watch` so that if you pass an array then that completely replaces the existing set of watched fields? Passing a single value would be continue to be purely additive as it is now. This would make it work more like `subscribe` currently does.

Seems good to me. It would be consistent with our subscribe behavior. A nice property of this approach is that your extension only needs to track the set of things it wants to watch rather than tracking what it is already watching and having to call unwatch.

defunctzombie, 2022-06-08:
> * Parsing large topics for `allFrames` should not crash Foxglove Studio

Unfortunately this is "tricky" due to browser memory limitations that vary across operating systems. We have some control over how much memory we are using but on different linux distributions the amount of memory available in the browser instance (which is also true of the desktop app) is more strict than we've observed on MacOS. Since other parts of the app are also using JS memory at some point you will have used up the available memory in the JS browser environment. Trying to subscribe to allFrames for large topics has this effect (since they all need to be kept in memory) and I would generally say your mileage with doing this will vary.

There are a few APIs we might be able to use to tell us how much memory is in use but as I've noted we have observed different behavior on Linux vs MacOS on whether Studio actually crashes vs continues to work and has more memory to use.

defunctzombie, 2022-06-08:
I'm removing the bug label on this since the behavior is OS specific and generally when you've used up the memory the browser allows for javascript its up to the browser implementation on that OS to determine what to do (linux likes to crash the browser instance).

Separately from using up memory - it sounds like you'd like the ability to _unwatch_ render state fields?

f-o-w-l, 2022-06-13:
> I think we're going to add another signature to `context.watch` that lets you alternatively pass an array instead of a single property and semantically if you pass an array then we treat that as the complete set of properties you want to watch and we'll unwatch the others. Would that work?

For sure, if the unwatching is happening, I'll make whatever work!

jtbandes, 2022-06-13:
> Basically you're saying here that the cross-platform Electron libraries that ya'll have tried for querying memory usage have turned up false values on some Linux distros? That's unfortunate.

I'm not sure we've tried querying memory usage from the app itself. We've just noticed that the app crashes on Linux in scenarios where it doesn't seem to crash as much on macOS.

defunctzombie, 2022-06-14:
@f-o-w-l It seems like the conclusion to this issue is a feature request to "unwatch" `allFrames`. Ideally the app wouldn't crash but as we've noted that seems to be somewhat outside of our control on linux which doesn't like giving much memory to browser apps.

I'd like to change this from a bug to a feature request given what we've discussed here.

defunctzombie, 2022-06-30:
@f-o-w-l while we haven't added an unwatch feature, I have landed some improvements to the experimental player which might help with this (depending on the specific use case). These changes are available in the web version now and will be in next week's desktop release.