foxglove/studio #3480: Support for protobuf `oneof`

jtbandes, 2022-06-01:
The protobuf support in `@foxglove/mcap-support` (https://github.com/foxglove/studio/blob/e5d1f84914d9d63ce4ae032d40807b4229b696df/packages/mcap-support/src/protobufDefinitionsToDatatypes.ts#L42) should support messages that use `oneof`.

amacneil, 2022-06-02:
I assume this will make message path autocomplete a little crazy if people do this right?

I think we should still support it either way.

jtbandes, 2022-06-02:
I think we would probably treat it the way the protobuf generated code does, where the oneof is like a message with a bunch of optional submessages for each alternative (kind of like our ImageAnnotations / Markers container messages, but with single items instead of arrays).

defunctzombie, 2022-06-13:
We believe this works until we learn otherwise with a specific example.

closed 2022-06-13 by defunctzombie