foxglove/studio #3487: Custom map tile url

OgreTransporter, 2022-06-03:
**Description**

I cannot set a custom tile url. No requests arrive on the local server.

[image]

- Version: 1.13.2
- OS: Windows 10
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): nuScene Demo
- ROS distro (if applicable):

**Steps To Reproduce**
Installed Apache and copied rendered tiles for levels 16 to 20 into htdocs folder. Set the port to 8043. In foxglove set the url to `http://localhost:8043/{z}/{x}/{y}.png`.

**Expected Behavior**
I expected the tiles to be loaded from the local server.

**Actual Behavior**
No requests arrive on the server - no matter which file. If I use the standard example of leaflet in a simple html file it works.

jtbandes, 2022-06-03:
Can you open up the developer tools (press alt key → View → Advanced) and see if any additional warnings or errors are displayed there?

OgreTransporter, 2022-06-07:
```
warn.js:14 Some icons were re-registered. Applications should only call registerIcons for any given icon once. Redefining what an icon is may have unintended consequences. Duplicates include: 
Add, AddIn, ArrowCollapse, ArrowCollapseUp, ArrowDown, ArrowExpandUp, ArrowLeft, ArrowLeftRight, ArrowRight, ArrowStepLeft (+ 107 more)
warn @ warn.js:14
PanelExtensionAdapter.tsx:403 [..\..\packages\studio-base\src\components\PanelExtensionAdapter.tsx] Init panel 879efc58-ce2d-445c-8b8b-000e4e1098ef
index.tsx:121 [..\..\packages\studio-base\src\providers\CurrentLayoutProvider\index.tsx] Panel action resulted in identical config: Object
(anonymous) @ index.tsx:121
PanelExtensionAdapter.tsx:403 [..\..\packages\studio-base\src\components\PanelExtensionAdapter.tsx] Init panel d2f2cc4d-048a-4d20-a198-7071f1986e00
index.tsx:121 [..\..\packages\studio-base\src\providers\CurrentLayoutProvider\index.tsx] Panel action resulted in identical config: Object
(anonymous) @ index.tsx:121
CachedFilelike.ts:57 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Opening file with size 486.557MiB
CachedFilelike.ts:130 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Setting new connection @ 0-200MiB
CachedFilelike.ts:130 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Setting new connection @ 486.318-486.557MiB
CachedFilelike.ts:135 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Destroyed current connection @ 0.014-200MiB
CachedFilelike.ts:203 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Connection @ 486.556-486.557MiB finished!
CachedFilelike.ts:130 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Setting new connection @ 0.014-200.004MiB
warn.js:14 Some icons were re-registered. Applications should only call registerIcons for any given icon once. Redefining what an icon is may have unintended consequences. Duplicates include: 
Add, AddIn, ArrowCollapse, ArrowCollapseUp, ArrowDown, ArrowExpandUp, ArrowLeft, ArrowLeftRight, ArrowRight, ArrowStepLeft (+ 224 more)
warn @ warn.js:14
PanelExtensionAdapter.tsx:403 [..\..\packages\studio-base\src\components\PanelExtensionAdapter.tsx] Init panel d2f2cc4d-048a-4d20-a198-7071f1986e00
index.tsx:121 [..\..\packages\studio-base\src\providers\CurrentLayoutProvider\index.tsx] Panel action resulted in identical config: Object
(anonymous) @ index.tsx:121
index.tsx:121 [..\..\packages\studio-base\src\providers\CurrentLayoutProvider\index.tsx] Panel action resulted in identical config: Object
(anonymous) @ index.tsx:121
index.tsx:121 [..\..\packages\studio-base\src\providers\CurrentLayoutProvider\index.tsx] Panel action resulted in identical config: Object
(anonymous) @ index.tsx:121
index.tsx:121 [..\..\packages\studio-base\src\providers\CurrentLayoutProvider\index.tsx] Panel action resulted in identical config: Object
(anonymous) @ index.tsx:121
index.tsx:121 [..\..\packages\studio-base\src\providers\CurrentLayoutProvider\index.tsx] Panel action resulted in identical config: Object
(anonymous) @ index.tsx:121
warn.js:14 Some icons were re-registered. Applications should only call registerIcons for any given icon once. Redefining what an icon is may have unintended consequences. Duplicates include: 
Add, AddIn, ArrowCollapse, ArrowCollapseUp, ArrowDown, ArrowExpandUp, ArrowLeft, ArrowLeftRight, ArrowRight, ArrowStepLeft (+ 107 more)
warn @ warn.js:14
CachedFilelike.ts:203 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Connection @ 200.002-200.004MiB finished!
CachedFilelike.ts:130 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Setting new connection @ 200.004-400.001MiB
CachedFilelike.ts:203 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Connection @ 400-400.001MiB finished!
CachedFilelike.ts:130 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Setting new connection @ 400.001-486.557MiB
CachedFilelike.ts:203 [..\..\packages\studio-base\src\util\CachedFilelike.ts] Connection @ 486.557-486.557MiB finished!
DevTools failed to load source map: Could not parse content for file:///C:/Users/transporter/AppData/Local/Programs/foxglove-studio/resources/app.asar/renderer/preload.js.map: Unexpected end of JSON input
DevTools failed to load source map: Could not parse content for file:///C:/Users/transporter/AppData/Local/Programs/foxglove-studio/resources/app.asar/renderer/module.wasm.map: Unexpected end of JSON input
```

The access logs from Apache are completely empty, i.e. not a single tile request was sent. For testing I opened single tiles directly in the browser and tried leaflet examples. Everything worked. In the browser version of foxglove it works too.

OgreTransporter, 2022-06-09:
I could now force an error message:

> Refused to load the image 'http://localhost:8043/18/206645/130128.png' because it violates the following Content Security Policy directive: "img-src 'self' data: https: package: x-foxglove-converted-tiff:".

OgreTransporter, 2022-06-09:
If I start foxglove in the browser like you do, it works for me too. Only the stand-alone Windows version doesn't work (I haven't tested macOS and Linux). Maybe it's just a setting error of bundled Chrome.

PR #3594 (2022-06-14): Add http to img-src CSP for desktop
**User-Facing Changes**
Allow using `http://localhost` for custom map tile urls in the map panel when using the desktop app.

**Description**

We want to allow loading images from http urls (i.e. http://localhost) to support users with local map tile servers. This works in the web version because browsers allow requests to http localhost from https pages.

Fixes: #3487
commit b0a70eda (2022-06-14): Add http to img-src CSP for desktop (#3594)

We want to allow loading images from http urls (i.e. http://localhost) to support users with local map tile servers. This works in the web version because browsers allow requests to http localhost from https pages.

Fixes: #3487
closed 2022-06-14 by defunctzombie