foxglove/studio #3490: Extensions should be able to publish internal messages

rgov, 2022-06-03:
I believe messages published by the Node Playground are "internal" to Studio in that they can be seen by other panels but don't actually get sent out over the network.

How about an extra flag to `PanelExtensionContext.publish()` that allows an extension to do the same?

Use case: I was thinking of a potential way to support video codecs; I could have an extension that decodes the video messages and then internally re-publishes the decoded image so that an Image panel can render it, with zoom support and markers and everything else "for free".

Related to #2212.

amacneil, 2022-06-03:
This seems like a duplicate of https://github.com/foxglove/studio/issues/2212? How is it different?

jtbandes, 2022-06-04:
To clarify, I think our plan is for each extension package to be able to implement more than one _extension point_. For example, today, there is no reason one extension can't provide 5 new kinds of panels via multiple calls to [registerPanel](https://github.com/foxglove/studio/blob/5680d79476a20e4ccc1889a1c52d06ddfcce80d4/packages/studio/index.d.ts#L246-L251). But I can understand that it would be useful for user interactions with the panel to be able to change/redo the message transformations, so we'll need to keep in mind that extensions using multiple extension points may have side effects between their extension point.

rgov, 2022-06-04:
That sounds great, but in the case of the GeoJSON example, you're not really transforming anything, you're purely synthesizing messages.

A fair question is also: Why not just publish regular messages? Except now Studio is sending out messages that potentially wind up in a rosbag or mcap unless you are very careful with your exclusion regex.

amacneil, 2022-06-06:
> A fair question is also: Why not just publish regular messages? Except now Studio is sending out messages that potentially wind up in a rosbag or mcap unless you are very careful with your exclusion regex.

This is an interesting question, I don't think there is a first principles answer other than that is the way it currently works. Perhaps we should make node playground and transform extensions always publish back to the live data source (although not for log files). One challenge is that we would be limited to message types where we have an existing schema (e.g. you wouldn't be able to publish arbitrary json).