foxglove/studio #3491: ROS2 connected but data is not showing

Floheile, 2022-06-03:
**Description**
Not all data from ROS2 is not showing in foxglove. The topic names are read, but the data e.g. ROS2 demo talker node is not shown.

- Version: Foxglove Studio version 1.13.1
- OS: Ubuntu 18.04 on Nvidia Jetson Xavier with Chromium Browser
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS2 Native
- ROS distro (if applicable): ROS2 dashing (currently required for Intel realsense package)

**Steps To Reproduce**

started ROS2 demo talker node

**Expected Behavior**
displaying the talker node's message count

**Actual Behavior**
foxglove studio connects but does not show the data. The talker node messages are not received/shown. Im currently also using an Intel Realsense Camera and would like to display the image stream. The camera info messages are arriving. Interestingly, in rare/ not reproducible occasions single camera image frames are showing. Nevertheless none of the demo talker node messages have been received. 
[image]

Am I doing something wrong in using it with ROS2 ?
Thanks for any advice

jhurliman, 2022-06-03:
The ROS 2 native connector has not been tested with Dashing, there are likely compatibility issues with the DDS RMW in Dashing. I will try to reproduce this if I have some time.

defunctzombie, 2022-06-13:
We align our support in Studio to maintained ROS releases. Dashing is considered end-of-life and we won't be supporting it in Studio. I'd recommend trying to upgrade your ROS release.

closed 2022-06-13 by defunctzombie