foxglove/studio #3496: ROS node error when I add turtle1/pose to RAW message panel

LeZhang13, 2022-06-06:
**Description**

- Version:1.12.1
- OS:Ubantu 18.04 (Virtual Machine）
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):ROS 1 native
- ROS distro (if applicable): turtlesim

**Steps To Reproduce**

Step by step with the Blog "Running Your First ROS Node on Ubuntu, macOS, or Windows"
Connect Foxglove to  ROS node turtlesim
Add [Topic Graph] panel to see Topic and node , this step is OK.
[image]
Add [Raw Messages] panel and add /turtle1/pose to see the structure of the topic's messages: NOK
[image]

**Expected Behavior**
To see the turtle1/pose in RAW message

**Actual Behavior**
ROS Node Error happened.
Error: Subscribing to /turtle1/pose at http://tedzhang-virtual-machine:44671/ failed (Error: requestTopic("/turtle1/pose") from http://tedzhang-virtual-machine:44671/ failed. err=Error: XML-RPC call "requestTopic" to http://tedzhang-virtual-machine:44671/ returned 502: "Bad Gateway"), this connection will be dropped
    at RosNode._subscribeToPublisher (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:26617:32)
    at async file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:26577:69

foxymiles, 2022-06-06:
What did you set as your ROS_HOSTNAME when you connected to the ROS1 source?

[image]

LeZhang13, 2022-06-07:
Hi, Same as yours
[image]

LeZhang13, 2022-06-07:
By the way , I also try to use ROS  native connect to play rosbag.
1. roscore
2. Foxglove connect ROS 1(same as last picture)
3. rosbag play xxx.bag
4. Foxglove show error
[image]
Also show bad gateway

LeZhang13, 2022-06-08:
> What did you set as your ROS_HOSTNAME when you connected to the ROS1 source?
> 
> [image]
HI, foxymiles, I found all ros native connect will show error : bad gateway .
I make a publisher to show my web camera video, and connect foxglove,  also show same error.
[image]
and
[image]
and 
[image]

defunctzombie, 2022-06-08:
These are ROS1 network setup (and env/hostname) resolution errors. Your nodes are reporting their hostname as `tedzhang-virtual-machine` however your desktop (where you run studio) is either not able to resolve that hostname or the port is being blocked.

Using ROS1 across hosts (even with virtual machines) requires that the ROS_HOSTNAME that all nodes are broadcasting to the master for themselves be resolvable from all other nodes that want to subscribe. It also requires full and open TCP network connectivity between all these nodes (and Studio which is just another node).

I recommend reading http://wiki.ros.org/ROS/NetworkSetup and making sure all your nodes have a correct ROS_HOSTNAME set that is resolvable across all nodes and studio (consider using an IP address rather than hostname to be sure).

Closing this because the issue is with network and environment setup and is expected behavior when trying to use ROS1 across hosts.

LeZhang13, 2022-06-10:
> These are ROS1 network setup (and env/hostname) resolution errors. Your nodes are reporting their hostname as `tedzhang-virtual-machine` however your desktop (where you run studio) is either not [...]

Thanks for your reply, but my foxglove studio and ROS all in my Ubantu vertical machine environment, I will check my Ubantu firewall.