foxglove/studio #3502: Desktop: Unable to open data source modal once sample data is loaded

esthersweon, 2022-06-06:
**Description**
In the desktop app, the plus icon that opens the data source modal (in the sidebar's "Data source" tab) disappears once I load sample data.
If I refresh the window, the sample data still loads, but the plus icon is displayed correctly.

- Version: 1.13.1
- OS: macOS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Sample data
- ROS distro (if applicable): N/A

**Steps To Reproduce**

Open desktop app, select "Explore sample data", and open the "Data source" tab in the sidebar.

**Expected Behavior**
Plus icon should be present in top right corner of sidebar.

**Actual Behavior**
Plus icon has disappeared.

defunctzombie, 2022-06-08:
The only way I see that icon not showing is if you've somehow got the OPEN_DIALOG app setting off:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/components/DataSourceSidebar/DataSourceSidebar.tsx#L86

I don't think there's even a UI for it and it certainly should change between opening the dialog. I've not been able to reproduce the behavior described.

Could you post a video of it happening?

esthersweon, 2022-06-15:
@foxymiles - I followed your advice of deleting and recreating the "Sample: NuScenes" layout, and I haven't been able to reproduce the issue after that. Maybe my original nuScenes layout had something that was causing the issue? Whatever it was, it seems to be gone now.

The only other change I made since reporting this bug was updating the app to v1.13.2.

defunctzombie, 2022-06-15:
Closing this since it sounds like no one can reproduce anymore. If we are able to reproduce we can revisit and fix.

closed 2022-06-15 by defunctzombie