foxglove/studio #3505: Image panel - fix image annotation topics sort order

defunctzombie, 2022-06-06:
**Description**
When the image annotation topics were in the toolbar as a dropdown we would sort "relevant" topics before other topics so they appear first in the list. Relevant topics are those matching the image topic prefix.

[image]

With the new annotation topics list in panel settings this list is unsorted making the relevant topics harder to find.

amacneil, 2022-06-06:
Besides updating the new sidebar with the improved sort order, it would be nicer still if we could visually separate "recommended" vs "other" (non official terms) annotation topics. E.g. a menu separator:

[image]

amacneil, 2022-06-06:
Related - https://github.com/foxglove/studio/issues/3509

PR #3513 (2022-06-07): Put related annotation topics at top of annotation list in image panel settings
**User-Facing Changes**
This sorts the annotation topics in the image panel settings so that topics related to the camera topic are at the top of the list.

**Description**
This also changes the styling of the in-panel camera topic picker to a standard select since it's no longer a multi-valued select.

[image]

Note that this does not implement annotation separation / segmentation as described in #3505 since it seems we need to think through the right approach in more detail first.

Fixes #3505
closed 2022-06-07 by foxymiles