foxglove/studio #3535: allFrames stopped working as of v1.13.2, working on v1.13.1

f-o-w-l, 2022-06-08:
**Description**
As of the latest version v1.13.2, watching `allFrames` is broken.

For example, with a `.bag` file with **63** messages on a given test topic, `allFrames` finds **0** if the `.bag` file is not played initially to trigger buffering. If the user seeks to halfway through the `.bag` to trigger buffering, then uses an extension to subscribe to a topic after watching `allFrames`, `allFrames` only finds either **0 or 2** messages on my test topic, between ~5 tests. Waiting does not help, it is never updated to contain additional messages.

On version v1.13.1, if the `.bag` is not played initially to trigger buffering, `allFrames` consistently finds **13/63** messages between multiple tests. If the user seeks to halfway through the `.bag` to trigger buffering, then it consistently finds `63/63`.

The accumulated data on my 63-message test topic is not anywhere close to 1GB.

- Version: 1.13.2
- OS: Pop!_OS 20.04 LTS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag file
- ROS distro (if applicable): N/A

**Steps To Reproduce**
Described above.

**Expected Behavior**
`allFrames` should consistently find all messages on a given topic with accumulated messages less than 1GB of data.

**Actual Behavior**
`allFrames` is populated with hardly any or no messages, after `context.watch("allFrames")` then `context.subscribe("/some/topic")` is called.

f-o-w-l, 2022-06-08:
> Can you check the "experimental bag player" setting in the app settings? We changed the default value of this setting (from on to off) in 1.13.2, so I'm wondering if that's what caused this. You can manually override the setting in the sidebar.

I've confirmed that this is the issue. After enabling the "Experimental bag player" and restarting Foxglove Studio, `allFrames` is back to working with only its normal inconsistencies. Thanks for the quick response!

defunctzombie, 2022-06-08:
@f-o-w-l to confirm - allFrames is _working_ or _not-working_ for you with the "Experimental bag player"?

defunctzombie, 2022-06-30:
@f-o-w-l Following up on this since I've landed some improvements to the experimental bag player. Those changes are available on web now and will be in desktop next week. While the player is "experimental" I'd recommend giving it a spin and seeing if it resolves your issue here. I'm moving towards that being the default player and will consider this bug fixed if its working for you in the experimental player.

defunctzombie, 2022-07-11:
Closing this out since we landed some improvements and I've not heard back.

closed 2022-07-11 by defunctzombie