foxglove/studio #3555: Propagate settings field error status through tree

rgov, 2022-06-09:
Foxglove Studio 1.13.2

If one of my settings fields has an error, but the tree is collapsed, I won't know about it.

It would be nice if the parents showed the error icon and clicking on it revealed the erroring field.

defunctzombie, 2022-06-12:
@foxymiles @2metres sending your way for UX ideas and implementation.

jtbandes, 2022-09-08:
I believe our latest thinking is that this would pose a performance problem. Is that right @foxymiles?

foxymiles, 2022-09-08:
Right. I can't think of a general solution to this problem that doesn't require recursing through all descendants on each render. There might be a pure CSS solution using `has()`.

jtbandes, 2022-09-08:
Did we have other ideas for how to address the UX issue of not being able to find errors?

foxymiles, 2022-09-08:
At the moment there's nothing stopping panel authors from handling this themselves explicitly. If we want to solve it in a general way I'd be tempted to try some CSS based `has()` solutions.

defunctzombie, 2022-09-19:
> performance killing tree traversal on every render on the settings UI side

Only on every update to the settings state not every render?

foxymiles, 2022-09-19:
> Only on every update to the settings state not every render?

You mean every render of the parent panel? There are many cases where settings gets updated very rapidly like panning the camera in 3d panel or scrubbing a number value or the color picker or even typing quickly in a settings field.

I still think trying something with the `has:` selector is worthwhile.