foxglove/studio #3567: Plotting quaternions

imranlivt, 2022-06-10:
The orientation in most of ros messages is of type quaternions. It doesnot make sense to plot x,y,z,w values. Is there any way I can convert the quaternion to euler angles, roll, pitch, yaw and plot them in foxglove. It seems to be qn important feature , yet i found no information regarding that. Am i missing something

rgov, 2022-06-10:
You could probably use a Node Playground script to implement a transformation to roll, pitch, yaw.

According to [this StackOverflow answer](https://stackoverflow.com/a/64874930/145504) the calculation should be like this:

```
yaw = Math.atan2(2.0*(qy*qz + qw*qx), qw*qw - qx*qx - qy*qy + qz*qz);
pitch = Math.asin(-2.0*(qx*qz - qw*qy));
roll = Math.atan2(2.0*(qx*qy + qw*qz), qw*qw + qx*qx - qy*qy - qz*qz);
```

defunctzombie, 2022-06-12:
We're happy to add a function to the node playground built-in utilities or examples if someone wants to take the initiative and write one up.

https://github.com/foxglove/studio/tree/main/packages/studio-base/src/players/UserNodePlayer/nodeTransformerWorker/typescript/userUtils