foxglove/studio #3570: Large point cloud (or medium point cloud with a long decay) support

jhurliman, 2022-06-10:
We need to identify the bottlenecks in rendering point clouds over a long decay period, where tens of millions of points accumulate. There may be some initial wins by optimizing draw calls or avoiding repeated work. We may also need to move to a more advanced rendering technique for large clouds such as https://github.com/potree/potree/

siddancha, 2022-06-12:
**ROS Bags for Benchmarking**

Here are some benchmark bag files. They contain poinclouds of the [Stanford Lucy model](http://graphics.stanford.edu/data/3Dscanrep/) containing 140,278,72 points. The following bag files contain `PointCloud2` messages on topic `/lucy` in the `map` frame:

1. [One large PC message (803 MB)](https://drive.google.com/file/d/1VakjBgs0IG7iHjnGO7YfLlL2WhWISNdH/view?usp=sharing): contains a single large message with 5 replications of Lucy (~701K points). This is near the limit of what Foxglove can load in a single message.
2. [Many medium PC messages (3.2 GB)](https://drive.google.com/file/d/1CuRmFOmZ-Kt5IxbSIAiGql4fPDbnuij7/view?usp=sharing): contains 20 messages with un-replicated Lucy (~140K points each). Foxglove will crash if you set a decay high enough that more than 5-6 messages are accumulated.

defunctzombie, 2022-06-12:
Thank you for sharing these!