foxglove/studio #3574: Ability to save plots as images

jameskuszmaul-brt, 2022-06-10:
It would be nice to have an option to export the current state of a Plot panel as an image. PNG would probably be adequate for most uses, although I know I've appreciated having SVG (or other vectorized formats) available for export in the past for inclusion in higher-quality reports.

rgov, 2022-06-10:
There's a [`toBase64Image()`](https://www.chartjs.org/docs/2.9.4/developers/api.html#tobase64image) API for this. A few years ago someone [tinkered with exporting to SVG](https://stackoverflow.com/questions/45563420/exporting-chart-js-charts-to-svg-using-canvas2svg-js), it might work.

amacneil, 2022-06-11:
More generally, would be cool if you could save any panel (or the whole layout) as a png. Currently only image panel supports it.

defunctzombie, 2022-06-12:
> More generally, would be cool if you could save any panel (or the whole layout) as a png. Currently only image panel supports it.

I am a little /shrug on this. I personally find it trivial to take a screenshot. I think I feel similarly about saving the plot to an image. SVG could be compelling (for the higher quality reports case), but I kinda feel like the answer there might be to fullscreen the panel, take a screenshot, and use that in a report?

defunctzombie, 2022-06-12:
> Currently only image panel supports it.

Image panel supports it because it actually gets you the original data not a screenshot of the data. This is similar to the CSV download of the plot data.

amacneil, 2022-06-12:
One thing to keep in mind is that taking screenshots isn't quite so trivial on windows/linux. It's possible, but can be quite frustrating.

jameskuszmaul-brt, 2022-06-13:
Taking screenshots gets the pixels across (and is what makes this not too high a priority in my mind), but there's a real jump in quality/convenience to be able to grab images that already have perfect cropping and don't require fiddling with the system screenshot tools. There're also some other benefits if you can provide vectorized or higher-than-screen resolution export options, although that is also generally more work. Obviously if you have very high standards for your graphics, you are probably exporting the CSV and generating a bespoke plot using other tooling anyways. But I don't think that's all that common a use-case.

rgov, 2022-06-13:
For high quality printed screenshots or presentations, don't forget you can increase the UI scale with <kbd>⌘+</kbd>/<kbd>⌘–</kbd> (or platform equivalent) to render the panel larger, then use an image editor to change the DPI.