foxglove/studio #3606: GLIBC_2.29 not found

osholsonel20, 2022-06-14:
**Description**

- Version: 1.14.2
- OS: Ubuntu 18.04 arm64
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS 1
- ROS distro (if applicable): Melodic

**Steps To Reproduce**

- Download the debian file on a Ubuntu 18.04 arm device.
- Run foxglove-studio

Previous version of Foxglove successfully ran, now after the update it no longer does on this specific device combination

[image]

defunctzombie, 2022-06-14:
I suspect this is caused by the upgrade of Electron: https://github.com/foxglove/studio/commit/f161057f53e6b4e512b29116589327ca0ca056f4

defunctzombie, 2022-06-14:
Electron identified the issue and has a fix: https://github.com/electron/electron/pull/34491

This fix is not yet in a release of electron for v19 but based on it merging into the v19 backport branch should be in v19.0.5

amacneil, 2022-06-14:
In the meantime, we keep old versions of Foxglove in our apt repo, so you should be able to downgrade with:

```sh
sudo apt install foxglove-studio=1.13.2
```

osholsonel20, 2022-06-21:
Just kind of a heads up here, I removed electron 19.0.3, and called "yarn add electron@npm:19.0.5". Built the debian then tested on an Arm64 Ubuntu 18.04 device and it's working as intended! 19.0.5 was just recently added to npm https://www.npmjs.com/package/electron

PR #3656 (2022-06-21): Upgrade electron to v19.0.5

**User-Facing Changes**
The desktop app will work on Ubuntu 18.04 arm64 systems.

**Description**
v19.0.3 regressed arm64 support for Ubuntu 18.04. This new releases adds back support for arm64 on ubuntu 18.04.

Fixes: #3606
commit d1ef0951 (2022-06-21): Upgrade electron to v19.0.5 (#3656)

v19.0.3 regressed arm64 support for Ubuntu 18.04. This new releases adds back support for arm64 on ubuntu 18.04.

Fixes: #3606
closed 2022-06-21 by defunctzombie