foxglove/studio #3613: Unable to play large rosbag file stored in azure blob from specific start and end time without downloading

saumilsdk, 2022-06-15:
Is there a way to play large rosbag file stored in azure blob from specific start and end time without downloading?

defunctzombie, 2022-06-15:
We support remote files via the remote file connection option: https://foxglove.dev/docs/studio/connection/remote-file. There are [nuances to getting remote files working](https://foxglove.dev/docs/studio/connection/remote-file#setting-up-cors-for-a-remote-file) and those might be different on different cloud providers. The easiest way to use remote files is to upload them to your foxglove account. We deal with all the gotchas of organizing and making the files streamable with specific start/end times so you only download the data you are visualizing.

For general questions I'd recommend our discussion board (https://github.com/foxglove/studio/discussions) or our slack channel. If this is a bug please provide steps to reproduce the issue.

defunctzombie, 2022-06-15:
If you are using the browser version of studio take a look at these docs: https://foxglove.dev/docs/studio/building-and-sharing-links some data sources allow you to specify a `time` url argument which will seek to that specific time within the data source.

saumilsdk, 2022-07-29:
@defunctzombie I have tried with the same but still it doesn't seem to work.
E.g.
https://olamainqoqg5.blob.core.windows.net/rawdata/Duster/2022-07-20/Duster__all-sensor_2022-07-20-11-16-27.bag?time=2022-07-20T11:16:50+05:30&se=2022-07-29T11%3A00%3A40Z&sp=r&sv=2021-06-08&sr=b&sig=XXXX