foxglove/studio #3615: Bulk save, revert, export, and delete layouts

jackiescanlon, 2022-06-15:
It would be awesome if we could bulk save, revert, export and delete layouts. Right now, the user must click the dot or three dots next to each layout and perform the action one at a time.

[image]

amacneil, 2022-06-16:
+1

Especially bulk delete would be super helpful

jtbandes, 2022-06-21:
cc @2metres – we would need some designs around how the selection should work. Or better, refactor this whole thing to use some Mui off the shelf component?

foxymiles, 2022-08-08:
@jackiescanlon All of these are implemented now except for bulk export which is a bit more complex. How important is bulk export to your workflow?

jackiescanlon, 2022-08-15:
Thanks so much! It would be useful at some point but having the other bulk actions is already a huge improvement.

amacneil, 2022-08-15:
Thanks for the feedback!

Closing this ticket since the most important bulk actions have been implemented.

Bulk export would require us to zip the exported layouts client side (since browsers can't reliably trigger multiple downloads at once), and I'm not sure that is worth the additional technical complexity, especially since exporting layouts is really only used for debugging.

closed 2022-08-15 by amacneil