foxglove/studio #3625: Changing panel type results in "Loading panel settings"

amacneil, 2022-06-16:
**Description**
1. Open panel settings sidebar
2. Change the panel type
3. Panel settings gets stuck in "Loading panel settings" state

https://user-images.githubusercontent.com/637671/173993879-c025c2d7-ab81-4d56-b37d-9a9cc1d3e3e6.mov

foxymiles, 2022-06-17:
This one is also a very old bug.

foxymiles, 2022-06-17:
Looks like the issue here is that when we change a panel type we don't update the selected panel id. It still points to the old panel id.

amacneil, 2022-06-21:
@foxymiles your PR fixed the issue described, but now there is a new bug. If settings sidebar is closed, and I change the panel, it gets "selected".

I also keep getting stuck in panel-select mode now. Infuriatingly, this seems to happen more often when I start screen recording, but I can trigger it sometimes without that too. When it happens, this is pretty bad, and as far as I can see is new behavior. Let me know if you can repro, might be easier to revert https://github.com/foxglove/studio/pull/3637 while we figure it out.

https://user-images.githubusercontent.com/637671/174716301-f0a49bbd-9740-42d8-bc96-b66e651ae5c1.mov

PR #3637 (2022-06-20): Automatically select newly swapped panels
**User-Facing Changes**
None

**Description**
The issue is that when we swap a panel we don't update the selected panel id so the context's selected panel still points to the id of the previous panel that no longer exists and the settings sidebar interprets this as no settings. 

The solution in this PR is to automatically make the newly swapped panel the selected panel. Due to the way the reducer is set up we don't know the id of the new panel so we have to do a slightly hacky before and after diff to figure it out.

Fixes #3625
PR #3653 (2022-06-21): Only select swapped panel if original panel was selected
**User-Facing Changes**
None

**Description**
This fixes a secondary issue caused by the fix for #3625. The change here is to only select the new panel if the panel we're swapping out was already selected.

Fixes #3625
closed 2022-06-21 by foxymiles