foxglove/studio #3626: Foxglove studio to support imagezero compressed image playback

saumilsdk, 2022-06-16:
We want to record camera images with imagezero compression which is fast and effective. https://github.com/swri-robotics/imagezero_transport

Foxglove studio image panel is unable to play images recorded in image zero compression.

jtbandes, 2022-06-20:
Hi, thanks for reaching out. At first glance it does not appear that imagezero is a very popular format, nor actively maintained (the documentation on [cfeck/imagezero](https://github.com/cfeck/imagezero) states "The code is “early work in progress”, and neither the image format, nor the library API are fully outlined yet", and the last update to the repo was 9 years ago). So it is unlikely that this request will be prioritized on our roadmap anytime soon, especially above other requests like https://github.com/foxglove/studio/issues/537 and https://github.com/foxglove/studio/issues/1069.

If you are interested in implementing this yourself, our supported image encodings are implemented here: https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/Image/lib/decodings.ts

You can find some prior discussion and add your input to this issue about custom encoding support via extensions: https://github.com/foxglove/studio/issues/1069

As always, feel free to discuss with us further on GitHub or in [Slack](https://foxglove.dev/slack).

closed 2022-06-20 by jtbandes

saumilsdk, 2022-07-06:
@jtbandes as per my knowledge foxglove is able to play compressed images but those images are lossy. Are there any lossless compression available other than imagezero?

jtbandes, 2022-07-06:
Yes, PNG and WebP compression are supported. See the list at https://foxglove.dev/docs/studio/panels/image