foxglove/studio #3643: New3D: Hide camera images by default

amacneil, 2022-06-17:
**Description**
The camera images add quite a bit of noise to the 3D scene and probably are a less common need. I think we could leave them turned off by default.

[image]

jtbandes, 2022-06-23:
I'm sure we could come up with some heuristics so the app doesn't grind to a halt for users with large amounts of data. But I am not convinced that having the panel be blank by default would be a better interim solution.

Here are some ideas for heuristics if we decide that showing all topics is too much.
- Prioritize topics by data size, if we know the size, otherwise sort by schema in order to estimate how expensive it is to show topics.
  - e.g. if we have to choose between a Pose or a Marker and an Image message, we should drop the Image first since it's likely to be much more expensive to visualize.
  - Then show at most 20 topics by default in this priority order.
- If we didn't turn on all topics by default, show the user a message/popup saying something to the effect of "some topics are hidden, open settings sidebar to customize".
- Keep in mind that the same layout may be used with different data sources over time, or a single live data source that gains more topics. So we have to respond dynamically to new topics coming online. This might be tricky since we don't want to store the list of "seen topics" in the layout. We might need to keep careful track of which topics the user has manually toggled on/off, and only dynamically decide what to do for topics that have never been manually toggled.
- Or just don't do any of this, because it sounds complicated and fragile

amacneil, 2022-06-23:
I think heuristics are a good idea. I have enjoyed the fact that New3D shows something useful by default, rather than a blank screen (although I did propose we hide image topics by default ( https://github.com/foxglove/studio/issues/3643 ) , they seem overkill).

However, we do need to think about data platform / remote sources, and how we can ensure those heuristics don't lead to a lot of unnecessary data download. Maybe we should be more conservative with remote data sources, although that seems overly complex.

jhurliman, 2022-06-23:
Superseded by [3692](https://github.com/foxglove/studio/issues/3692)

closed 2022-06-23 by jhurliman