foxglove/studio #3648: Better TypeScript types for settings tree APIs/handlers

jtbandes, 2022-06-20:
All uses of `action.path` in the settings tree are not really type safe, and we end up using lodash's `set()` to update values in the tree, which is also not type safe.

With some more effort, we may be able to harden the typings on settings paths/actions if each panel provides a strong type for its settings tree, e.g.:

[image]

defunctzombie, 2022-06-20:
Would this impact the surface area of the API (renaming, etc) and be something we want to tackle before making the api public?

foxymiles, 2022-06-21:
@defunctzombie This is more of a nice to have for the future than something that Jacob or I have a clear idea how to practically implement right now.

jtbandes, 2022-06-21:
Prototyping: [playground](https://www.typescriptlang.org/play?#code/C4TwDgpgBAyhzAJYDsDmBnAKgJwhAMhAIYBmUAvFOsNiqlAD5TICuAtgEYTaNQeKoUwXhwD2ogDbFkvFsgDWyUQHdkAbQC6AbgBQO0JFjwkaLLggBVMABMiwCAAUiICaKLWAJAEk2YCQB4nYAALKAgAD3tka3QoOUUVdQ0AGigcPAA+Cig1dIgNMMiIaNi1HSgjBDozPEJSZJ0NcqgAfigAbygwO2CALigg4K0oADciCRYIfryoAF9m-s61AGkoFCh5CBBRMjyNfrgq0zyrW3snFzdPHz9-NQA6R8HU5ZS08xWNLNm1Te3d8zaPQGaCHEwYE42OyOZyudz+GaUeJKVRZSi5QGFKIxHLICAjbhNCptTrdEL9ai0NCaYZjCZTOIKFEyeYVA7GaqQs4wy7uby+AKaVJ5DK6YHgaDdXnWbJgznmU7Qi5w6z+TqoYrccaLKAkcQUmh0VIcIjYfqsTjcOapNimzbYdD9ABKEAAxqJsKrKUaOiNEOhEBwpP0xJJpLNvhkdEA)

However even with these strong types, TypeScript's narrowing doesn't work all that well: https://github.com/microsoft/TypeScript/issues/48378#issuecomment-1162132154

So another approach like replacing the path/value objects with DeepPartial state diffs might be more amenable to strong typing throughout the code.

jtbandes, 2022-06-21:
Deemed too little benefit for most panels, for too much work. Complex dynamic panels are likely to circumvent the strong types anyway, and for simpler panels the types only help a little bit.

closed 2022-06-21 by jtbandes