foxglove/studio #3655: Impossible to view arrays of messages with high frequency in Raw Message Panel `Auto` mode

KerstinKeller, 2022-06-21:
**Description**
As per default configuration, when you try to view a message in the Raw Message Panel in "Auto expand" = "Auto" mode (which is the default), the tree automatically collapses when a new message arrives.
This is not the behavior in "Auto expand" = "Off" or "Auto expand" = "On".
I am not sure that this is the intended behavior?

https://user-images.githubusercontent.com/13848742/174843646-ae6202b6-d77d-4418-bffa-8d0c8127d594.mp4

- Version: 1.14.2
- OS: Windows

**Steps To Reproduce**

**Expected Behavior**
No instant collapsing after expanding the message view.

KerstinKeller, 2022-06-21:
Ok, understood. However I am not convinced that this is the best user experience. 
If a user clicks to expand a list, I'd assume that this list is kept open, in order to be anayzed, whereas the "auto" value just gives me a good starting point (neither all expanded, nor all collapsed).
It also took me a while to realize that there are options to configure for this panel 😉 
But I guess this issue can be closed, then.

foxymiles, 2022-06-21:
Actually on taking a closer look we do already have some code in there to preserve the user's clicks but it's not working. Will fix.

foxymiles, 2022-06-21:
@KerstinKeller So there was actually an issue here that should be fixed now. Thanks for the bug report and thanks for the feedback on settings discoverability!

PR #3659 (2022-06-21): Fix raw message auto mode expanded fields logic
**User-Facing Changes**
Fixes an issue with the behavior of the `auto` expansion mode of the raw messages panel.

**Description**
We have some logic in place to respect any paths the user has manually expanded in the raw messages panel in `auto` mode but it was incorrectly determining if there were any such paths.

Fixes #3655
closed 2022-06-21 by foxymiles