foxglove/studio #3669: Slow image hz vs. actual source

osholsonel20, 2022-06-22:
**Description**
In Foxglove studio the hz rate for my image topic is hovering around 5 hz, but in rviz and in rostopic hz /camera_data its around 30hz.
Is there some throttling going on or a way to increase the rate? Is it a bottleneck of some kind?

Some further background we tested:

- Foxglove running on Drive AGX device with camera stream on the same device
- Foxglove running on a separate device (i9 CPU) streaming information from Drive AGX to device (This time the speed was only 1.5hz)

- Version: v1.15
- OS: arm64 - Ubuntu 18.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS 1 Native
- ROS distro (if applicable): Melodic

**Steps To Reproduce**

**Expected Behavior**
Normalized camera speeds in line with rqt_image_view or rviz

**Actual Behavior**
Slower both running on device and on a different device

osholsonel20, 2022-06-22:
Also on the Drive AGX we get: ERROR:gpu_memory_buffer_support_x11.cc(44) dri3 extension not supported. Not sure if this would be relevant?

jtbandes, 2022-06-22:
Are you viewing the rate in the sidebar? Does the actual rate observed in the Image panel (or Raw Message panel) seem to be 5Hz?

osholsonel20, 2022-06-22:
Yes the sidebar listed is 5hz and the rate in the image also seems to be 5 hz / delayed

jtbandes, 2022-06-22:
Then you will need to figure out where the bottleneck is. You can try:

- Look at CPU vs network usage
- Try smaller images or other data sources to determine what the limit is
- Run the same software on a more powerful computer and see if it improves

osholsonel20, 2022-06-22:
We recorded a rosbag file of the camera... it seems 23.4 seconds creates 6 gigs of data....? It ran at around 30 hz on a powerful computer vs. the drive agx. That might be an issue? Would CompressedImage be a better topic to view? The camera is a sekonix 60° FoV 1920x1208 through GMSL. I wonder if its the data stream that's causing the huge bottleneck

osholsonel20, 2022-06-22:
So it seems our message for a raw image is 278 mb/s... 1920x1208 RGBA8 30 fps. I'm going to take a guess that we should compress the data or maybe even divide by 2 on h/w. Do you guys know of a "sweetspot" or some common image sizes / compressed vs. non compressed for streaming data in foxglove? Not recorded*