foxglove/studio #3670: [New3D] Camera image doesn't move  when change distance of camera info on setting panel

belopot, 2022-06-22:
**Description**
Camera image doesn't move  when change distance of camera info on setting panel

- Version:
- OS: Web
- Data source: Sample data

**Steps To Reproduce**
1. Open New 3D panel settings.
2. Change the distance of camera_info in Topics
[image]

**Expected Behavior**
The camera image should move to a new position
[image]

**Actual Behavior**
The camera image doesn't move to a new position
[image]

belopot, 2022-06-22:
@amacneil @jhurliman If you want. I will fix with myself

jhurliman, 2022-06-23:
This is working as intended. The purple frame and the image are two separate topics being visualized. You can change the distance of the image using the settings for the image topic.

closed 2022-06-23 by jhurliman