foxglove/studio #3682: Bug when switching between tabs

KerstinKeller, 2022-06-23:
**Description**
I am having an issue, where after switching tabs in a layout, no more data is being received.

https://user-images.githubusercontent.com/13848742/175323146-c284f2bf-17ed-420a-8e42-50f3277a7696.mp4

The steps I have taken (as can be seen in the video):
- Establish a websocket connection with a python websocket server, forwarding our data
- One tab is active, data is being sent
- I switch to another tab (contains a Map and 3D Panel)
- I switch back to the original panel, no more data is being received.

As can be seen from the server log, the first panel subscribes to channel 137. When switching tabs, that channel is unsubscribed and other channels are subscribed to.
When switching back, only the first channel is being unsubscribed (it's a marker array of the 3D panel), but the other channels are not being unsubscribed.

Also not that the duration of the DataSource does not count up any more.

I have also tried to reproduce with a simpler example:
-  Only 3D panel, no map, then switching back an forth works fine
-  Only map panel, or map panel together with raw message panel, switching forth and back works fine

I don't know what is special about this usecase, but I cannot fix it in any other way then to completely restart studio (plus the websocket server). I cannot be a 100% sure that it's not the websocket server (python library) that is causing the problems, maybe some race condition.

However, in this setup, this "crash" is 100% reproducible. 

I can record an mcap file to see if the behavior persists if replaying an mcap file.

- Version: 1.15.0
- OS: Windows
- Data source: Foxglove websocket connection (with Python server)

KerstinKeller, 2022-06-24:
Thanks for the update. 
Unfortunately, the fix doesn't resolve the problem. It still occurs. 
However, using the webapp (release build), it is happening far less than using the desktop app.

I think it must be related to the websocket communcation. I will try and record the websocket traffic, to see what's going on.
Hopefully that will give more insight.

KerstinKeller, 2022-06-24:
In the developer console I am also getting the following error trace when the bug happens:

```
index.tsx:68 Uncaught (in promise) 
_pendingCallbacks.<computed>	@	Rpc.ts:130
terminate	@	Rpc.ts:113
unregisterWorkerListener	@	WebWorkerManager.ts:69
(anonymous)	@	index.tsx:77
bj	@	react-dom.production.min.js:224
cj	@	react-dom.production.min.js:230
dk	@	react-dom.production.min.js:256
exports.unstable_runWithPriority	@	scheduler.production.min.js:18
gg	@	react-dom.production.min.js:122
Uj	@	react-dom.production.min.js:252
Lj	@	react-dom.production.min.js:243
(anonymous)	@	react-dom.production.min.js:123
exports.unstable_runWithPriority	@	scheduler.production.min.js:18
gg	@	react-dom.production.min.js:122
jg	@	react-dom.production.min.js:123
ig	@	react-dom.production.min.js:122
Jg	@	react-dom.production.min.js:237
Oh	@	react-dom.production.min.js:170
listener	@	index.ts:49
(anonymous)	@	index.tsx:51
listener	@	index.tsx:159
emit	@	index.js:202
(anonymous)	@	LayoutManager.ts:83
await in (anonymous) (async)		
(anonymous)	@	index.tsx:75
bj	@	react-dom.production.min.js:224
cj	@	react-dom.production.min.js:230
dk	@	react-dom.production.min.js:256
exports.unstable_runWithPriority	@	scheduler.production.min.js:18
gg	@	react-dom.production.min.js:122
Uj	@	react-dom.production.min.js:252
Lj	@	react-dom.production.min.js:243
(anonymous)	@	react-dom.production.min.js:123
exports.unstable_runWithPriority	@	scheduler.production.min.js:18
gg	@	react-dom.production.min.js:122
jg	@	react-dom.production.min.js:123
ig	@	react-dom.production.min.js:122
Jg	@	react-dom.production.min.js:237
Oh	@	react-dom.production.min.js:170
listener	@	index.ts:49
(anonymous)	@	index.tsx:51
listener	@	index.tsx:159
emit	@	index.js:202
(anonymous)	@	LayoutManager.ts:83
```
Not sure if it only relates to the problem, or causes the problem.

I tried to record the data going over the websocket, but have not really found my way around the developer tools, to acutally see all messages being sent over the websocket.

Also I remember that some time back in eCAL we did have multiple problems when e.g. destructing subscribers and such, in order to not run into problems with stale callbacks.

E.g you would always unregister your callback before you actually unsubscribe from a topic.
Maybe Studio doesn't handle it correctly, or there are still some edge cases in either server / client websocket library, e.g. when the Server sends a topic and the Client sends an unsubscribe request at the same time.

KerstinKeller, 2022-06-24:
I do have some more information, the issue must be located within the websocket protocol client / server libraries:
[image]
- Studio WS client sends all requests fine, when switching tabs
- it seems like the server gets disrupted sending a larg(er) message with the opcodes from the client

We can probably continue discussing this in the websocket library repo.
The server seems to get "hung up".
Maybe it can be recreated with a testcase of the server sending a large message and the client sending unsubscribing events at the same time.

jtbandes, 2022-07-01:
Sounds like this issue was determined to be a deadlock in the server implementation.

closed 2022-07-01 by jtbandes