foxglove/studio #3692: Hide all topics by default (matches behavior of old3d)

jhurliman, 2022-06-23:
For now let's focus on feature parity rather than changing the default topic behavior of the 3d panel. It is not obvious that we should be turning on all topics by default.

defunctzombie, 2022-07-11:
@jhurliman can you please make this change this week - it's an easy change and been sitting on the backlog.

defunctzombie, 2022-07-15:
Re-opening this issue because I am still seeing transforms on the screen when I open a new 3d (Beta) panel.

Here's a comparison of the current 3d panel and 3d (beta). I've opened the nuscenes sample dataset and added both panels to a new empty layout.

[image]

defunctzombie, 2022-07-15:
Spoke in slack. We agreed to leave these visible for now since 3d (beta) unconditionally subscribes to transform messages so there's no additional "cost" in showing these by default.

We also mentioned that having a grid in the scene by default would be a nice visual to orient the display and make it obvious its not broken if there's nothing else to render.

closed 2022-07-15 by defunctzombie