foxglove/studio #3705: Support linked variables in extension panels

jhurliman, 2022-06-24:


defunctzombie, 2022-07-12:
We need to make a call on whether we want this feature for GA.

jhurliman, 2022-07-22:
Summary from our discussion today on the path forward:

- Add two new methods to the panel extension API for reading the list of all global variables and setting a global variable as key + value, where value is a POD/POJO. (This API should probably closely mirror how parameters are handled in the extension API.)
- The 3D panel automatically calls `setVariable("selected", currentSelection)` whenever a renderable is selected, and `setVariable("selected", undefined)` on deselection. The `"selected"` variable name can be changed in panel settings.
- User Scripts have access to these global variables somehow. (JH: I'm not familiar with this aspect of global variables, is there a `getVariable("selected")` function in user scripts?)
- Message path syntax is updated to support sub-fields of global variables. For example, the currently selected object might be from the `/tracked_objects` topic, but I want to view `/planning.entities[:]{id==$selected.id}` in Raw Messages.

The net result is less user configuration to unlock this feature. You don't have to configure any variable linking, the functionality starts working as soon as you click on something in any 3D panel. The message path syntax for this is still a challenging learning curve, but we can provide copy-paste examples in the documentation now.

defunctzombie, 2022-07-22:
> * User Scripts have access to these global variables somehow. (JH: I'm not familiar with this aspect of global variables, is there a `getVariable("selected")` function in user scripts?)

User scripts get the variables as the second argument to their function. The first argument is the message event and the second is the global variables.

jhurliman, 2022-07-26:
The extension API change was merged in #3941. The change to New3D is in #3951 but with a lesser scope: a `selected_id` variable is used and only for markers. This obviates the need to change message path syntax right away.

closed 2022-07-26 by jhurliman