foxglove/studio #3707: Second picking pass for instanced renderables, point clouds

jhurliman, 2022-06-24:


jtbandes, 2022-06-25:
Why second pass instead of assigning an id per instance during the first pass?

jhurliman, 2022-06-25:
That could work if we think up a way to assign non-colliding IDs inexpensively.

defunctzombie, 2022-07-22:
Had a chat and going to push this until after GA.

This ticket is about selecting individual points in point clouds.

jtbandes, 2022-07-22:
Also applies to individual cubes/spheres in cube_list/sphere_list markers, right?

jhurliman, 2022-07-23:
Selecting cubes/sphere in lists works fine. You are shown the full marker message that contains the specific selection though, instead of drilling down to the array index for the specific cube or sphere you clicked on.

defunctzombie, 2022-08-17:
Re-surfacing this as higher priority. This is a feature of the existing 3d panel and is now a customer request. At minimum having the ability to click a point cloud point and see the x/y/z. We can re-target the issue for that specific feature if thats easier than the larger scope this issue might describe.

PR #4295 (2022-09-07): [New3D] Per-point point cloud and laser scan picking
**User-Facing Changes**

- Individual points in point clouds and laser scans can be selected in `3D (Beta)`

**Description**

[image]

This adds a pickInstance() method that is called to render a hitmap of instance indices for an instanced renderable. This allows individual points, lines, and primitives to be selected. Renderables can set `public override pickableInstances = true` and optionally specify a custom `userData.instancePickingMaterial` that should output `gl_FragColor` (or equivalent) equal to the 32-bit integer `instanceId` of th [... cut]
commit cb279225 (2022-09-07): [New3D] Per-point point cloud and laser scan picking (#4295)

This adds a pickInstance() method that is called to render a hitmap of
instance indices for an instanced renderable. This allows individual
points, lines, and primitives to be selected. Renderables can set
`public override pickableInstances = true` and optionally specify a
custom `userData.instancePickingMaterial` that should output
`gl_FragColor` (or equivalent) equal to the 32-bit integer `instanceId`
of the picked instance. If this is not specified, a default shader is
used that will use `gl_InstanceID`. Point clouds and laser sc [... cut]
closed 2022-09-07 by jhurliman