foxglove/studio #3714: Node playground topics should work in Plot panel

amacneil, 2022-06-25:
Node playground does not currently preload/backfill data when it is viewed in the plot panel. We should make it work.

[image]

jtbandes, 2022-06-25:
Related https://github.com/foxglove/studio/issues/2797

jameskuszmaul-brt, 2022-06-28:
I was about to ask for this! I look forward to whatever y'all figure out...

defunctzombie, 2022-06-30:
@jameskuszmaul-brt I made a POC you can try from the preview deployment of this PR: https://github.com/foxglove/studio/pull/3721 you can give it a try on local files.

I noted a possible gotcha in the PR that does alter what messages are fed into the scripts. I'll circulate this internally and see if we are ok with the change.

jameskuszmaul-brt, 2022-07-01:
I haven't actually had a chance to try this, but just in terms of some of the subtleties about how to handle non-monotonic preloading of data:
Supporting purely stateless node playgrounds is extremely valuable on its own, so I'll try to play with it in that context; however, I know a good chunk of the things I want to implement do require maintaining some state. E.g.:
1. Performing simple causal filtering on data.
   * E.g., calculating a simple first-order filter, or a basic window filter, or whatever.
   * Doing something like calculating the derivative of a signal would also fall into this (yes, I know the `.@derivative` already covers this particular case).
2. Non-causal filtering/processing on data (i.e., when you generate the output from your playground, you are able to either view future messages or specify a timestamp different than that of the input).
   * This would be a new feature relative to the existing playground.
   * This allows, e.g., doing a moving average that has no phase delay (or calculating an actually centered derivative).
3. Combining data between different topics published at similar rates (to, say, average a couple of different sensor values).
4. Cross-referencing data from a latched topic (or similarly data in the attachment section of an MCAP file) that can be used to interpret data in a periodic message (e.g., you have transform data for the position of a sensor in a latched topic and want to use it to reference sensor data to a common origin).

With (1), (3), and (4) in mind, I like the suggestion in https://github.com/foxglove/studio/pull/3721#issuecomment-1170670383 to create a separate instance of the script to handle separate ranges that may be loading--it will create some awkwardness on the boundaries, but should produce generally correct behavior. However, that also implies that once everything is loaded you will want to finish running whatever instance of the script was running from the "true" start, duplicating a bunch of work. I'm generally inclined to say that is an acceptable loss for many use-cases, but it could get irritating for stateless nodes. That sounds tractable with some sort of configuration, however.

defunctzombie, 2022-07-19:
@foxymiles Assigning to you since you are workin on this now. This is the issue related to the proof-of-concept PR.

commit a0e3445d (2022-07-19): Process preloaded messages with node playground

Node playground scripts are also run on preloaded messages. Panels like plot benefit from showing the entire loaded dataset when possible. Prior to this change, node playground scripts would only run "during playback" which meant that a plot panel could not show the output of a node playground script against the entire range of data.

This change alters this behavior by running the scripts on preloaded messages and playback messages.

Fixes: #3714
PR #3721 (2022-07-19): Process preloaded messages with node playground
**User-Facing Changes**
Adds support for plotting the outputs of node playground scripts.

**Description**
Node playground scripts are also run on preloaded messages. Panels like plot benefit from showing the entire loaded dataset when possible. Prior to this change, node playground scripts would only run "during playback" which meant that a plot panel could not show the output of a node playground script against the entire range of data.

This change alters this behavior by running the scripts on preloaded messages and playback messages.

This i [... cut]
commit 94d29638 (2022-07-19): Process preloaded messages with node playground (#3721)

* Process preloaded messages with node playground

Node playground scripts are also run on preloaded messages. Panels like plot benefit from showing the entire loaded dataset when possible. Prior to this change, node playground scripts would only run "during playback" which meant that a plot panel could not show the output of a node playground script against the entire range of data.

This change alters this behavior by running the scripts on preloaded messages and playback messages.

Fixes: #3714
closed 2022-07-19 by foxymiles