foxglove/studio #3731: Incorrect error message for mcap file with no messages

jtbandes, 2022-06-27:
When dropping this file into Studio, I get "The mcap file is unindexed. Only indexed files are supported."

[bad.mcap.zip](https://github.com/foxglove/studio/files/8996255/bad.mcap.zip)

However, the file is in fact indexed, so the error message is wrong.

[image]

defunctzombie, 2022-06-30:
When I try loading this same file in the non-experimental player I see this warning:

[image]

It also thinks the file is un-indexed.

Here is how we are identifying indexed files right now:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/IterablePlayer/Mcap/McapIterableSource.ts#L25

Is there something wrong with this logic? I copied it from the non-experimental player.

jtbandes, 2022-06-30:
Yeah, probably both are wrong. I guess they don't account for _empty_ files as a special case.

defunctzombie, 2022-06-30:
I can make a fix here if you think there is a fix we should make? Should we add logic to detect what we think to be an empty file? Or do we keep the existing error message?

PR #3768 (2022-07-06): Distinguish between un-indexed and empty mcap files

**User-Facing Changes**
Users of the experimental mcap player are given a more detailed error message if their mcap file cannot be opened with the indexed reader.

**Description**
Mcap files may have summary records but be missing chunkindex records (as a result of not having any messages). We distinguish between files with no summary information and those that might have some summary information but are missing the summary information we need.

Fixes: #3731

----
I'm open to other suggestion in the language here. A more robust check would b [... cut]
commit c4886147 (2022-07-06): Distinguish between un-indexed and empty mcap files (#3768)

**User-Facing Changes**
Users of the experimental mcap player are given a more detailed error message if their mcap file cannot be opened with the indexed reader.

**Description**
Mcap files may have summary records but be missing chunkindex records (as a result of not having any messages). We distinguish between files with no summary information and those that might have some summary information but are missing the summary information we need.

Fixes: #3731
closed 2022-07-06 by defunctzombie