foxglove/studio #3732: Misleading/meaningless message frequency shown for small mcap file

jtbandes, 2022-06-28:
Drop the following files into the app:

[array_of_bytes.mcap.zip](https://github.com/foxglove/studio/files/8996447/array_of_bytes.mcap.zip)

[array_of_bytes_unindexed.mcap.zip](https://github.com/foxglove/studio/files/8996448/array_of_bytes_unindexed.mcap.zip)

The indexed file shows a meaningless large number for the message frequency:
[image]

The unindexed file shows "Infinity Hz" with the non-experimental player (and is not supported with the experimental player):
[image]

foxymiles, 2022-07-12:
What do you think we should do in the first case?

jtbandes, 2022-07-12:
We shouldn't show a frequency if we can't compute one, so either show "–" or just show nothing

foxymiles, 2022-07-12:
In the first case we do have a message count and a timespan it just results in a crazy value since the timespan is so short. 

In the second case showing a dash seems like a good solution.

jtbandes, 2022-07-12:
Not sure what you mean – the two cases are the same (just indexed vs unindexed). Both of them contain a single message so the time range is zero. That shouldn't result in a random integer being shown for the rate – Infinity makes more sense mathematically since the denominator is zero, but I think it would be better to show a dash than Infinity.

foxymiles, 2022-07-13:
In the first case the message pipeline is calculating a non-zero time. Maybe that's the logic we need to fix?

closed 2022-07-19 by jtbandes