foxglove/studio #3739: Whether to support .H265 format video for rosbag?

lishulincug, 2022-06-28:
I use wasm to decode h265 frame from rosbag , but severe card and delay。

defunctzombie, 2022-06-28:
I don't understand the issue description. If you have a specific feature you'd like us to add please describe how you'd like it to work in more detail.

closed 2022-06-28 by defunctzombie

jhurliman, 2022-06-28:
HEVC decoding is not supported in Chrome (https://caniuse.com/?search=hevc), it would have to be done in WebAssembly as you mention and the codec is expensive to decode on the CPU. Video decoding will likely prioritize hardware-accelerated codecs.

lishulincug, 2022-06-28:
> 延迟
The latest chormium development version is already supported。
https://github.com/StaZhu/enable-chromium-hevc-hardware-decoding

lishulincug, 2022-06-28:
> [Chrome ( https://caniuse.com/?search=hevc](https://caniuse.com/?search=hevc) )不支持 HEVC 解码，正如您提到的，它必须在 WebAssembly 中完成，并且编解码器在 CPU 上解码的成本很高。视频解码可能会优先考虑硬件加速编解码器。

The latest chormium development version is already supported。
https://github.com/StaZhu/enable-chromium-hevc-hardware-decoding

itzhaolin, 2023-07-25:
i have same problem.

jtbandes, 2023-08-15:
See also: https://github.com/foxglove/community/discussions/794