foxglove/studio #3775: No indication that my mesh resource did not load in 3D (beta)

defunctzombie, 2022-07-05:
I've created an mcap file with markers that reference mesh resource urls. When I load my mcap file and visualize the marker topics in new 3d I don't see anything.

The behavior I'd expect is to see a topic error letting me know what happened when my markers couldn't load from their resource url.

siddancha, 2022-07-09:
Have you checked both the web and desktop versions?

This may be related to the following bug I came across: mesh marker messages with resource urls work on the web version but not on the desktop version. Here's a bag file that reproduces this bug. The marker topic is `/robot`: 
 [mesh_marker.bag.zip](https://github.com/foxglove/studio/files/9075807/mesh_marker.bag.zip)

jhurliman, 2022-07-14:
@siddancha thanks for the repro, fixed in https://github.com/foxglove/studio/pull/3838/commits/0eac7f192ec06229153c60255db2918c9edee273

defunctzombie, 2022-07-14:
My error was with the web version. The app tried to make network requests to `http://localhost:1234` which failed since I had no server running on localhost on the specific port in the resource url. I didn't get any indication this failure happened.

defunctzombie, 2022-07-14:
I've attached a screenshot to the linked PR - I don't think the URDF support PR needs to tackle this issue if it is ready otherwise but this issue remains unfixed by that PR for my use case. I failed to describe my original issue well - sorry about that. Hopefully the screenshot and above comment are clearer.

closed 2022-07-14 by jhurliman