foxglove/studio #3795: New3D panel doesn't show grid until data is available (frame error)

jtbandes, 2022-07-08:
Steps to reproduce:
- Load a data source.
- Add a grid.
- Reload the app and do not load a data source.

[image]

snosenzo, 2022-08-17:
what's the expected behavior here?

jhurliman, 2022-08-22:
The current New3D panel is built as a pipeline of datasource -> canvas, so no datasource = empty canvas. It would be nice to rearchitect the panel so rendering is somewhat decoupled from the datasource and we can render grids or other custom (user-added) layers when the app first starts.

I've reclassified this as `ux` following the triaging patterns we've adopted for other quirky behavior such as drag and drop of URDFs. It will be a nice UX improvement, although it's probably a heavy lift to implement.

defunctzombie, 2022-08-30:
Related: https://github.com/foxglove/studio/issues/4312

defunctzombie, 2022-11-04:
One quirk I've noticed when revisiting this is that I cannot select "<Display Frame>" from the Grid settings when I do not have any data loaded.