foxglove/studio #3799: Support ImageAnnotations without CameraCalibration

amacneil, 2022-07-08:
The ImageAnnotations still require a dummy (for the most part, except for height and width of the image) CameraCalibration. I think this can be better optimized by removing the dependency as it adds an extra field (CameraCalibration) that needs to be streamed/stored.

_Originally posted by @bhanug12 in https://github.com/foxglove/schemas/issues/21#issuecomment-1179179664_

jhurliman, 2022-07-08:
@bhanug12 what you are referring to as a "dummy" `CameraCalibration` likely means your image is rectified and you are supplying image annotations in rectified image space. This is just as meaningful as a CameraCalibration that tells us the original sensor had different pixel dimensions, or the images are unrectified and a distortion matrix is provided, or the sensor is binning pixels or the image is only a subset (region of interest) of the original sensor.

Since rectified images with matching image annotations are a common enough use case, this could be an option in Studio to select "None" for the CameraCalibration topic and just assume identity matrices with no distortion and get the image dimensions from the image topic.

bhanug12, 2022-07-12:
I see, did not know about use cases where ImageAnnotations would not be in rectified space. I assumed that processing needs to be handled before sending in data to foxglove. 
I agree that an option in Studio to default to None in CameraCalibration (maybe even use that as default) would be nice. 
I think this can also be mentioned in the docs for now - that sending in CameraCalibrations is required - in the ImageAnnotation page.