foxglove/studio #3806: Add title to indicator panel

bertaveira, 2022-07-09:
We use a lot of these panel to indicate status of systems but there is no way to distinguish them despite changing for example the "False" label to "System1 - False" which is not ideal.

Would be nice to be able to edit the title of the panel like you can with the plot panel.

jtbandes, 2022-07-20:
@2metres @foxymiles @defunctzombie what is our path to custom/editable panel titles? Is this something panels can already change using internal APIs and it just needs to be added to the extension API? Is custom names even something we want to support or should we just let panels put custom controls in the toolbar?

defunctzombie, 2022-07-22:
@jtbandes I see a few paths here:

* We expose some method on `PanelExtensionContext` and the panel logic can set the title.
* We make it possible to double-click the title to edit it and handle this automatically for all panels in the titlebar logic.
* Some combination of both of the above.

foxymiles, 2022-07-25:
@defunctzombie I'm inclined to option #2 but with a pencil icon to edit the panel title similar to how renaming nodes in settings works. Double click is non-discoverable and has interaction issues with things like selecting text.

foxymiles, 2022-07-25:
Also this isn't really documented but if we have a `title` field in the panel's config we display that in the title bar. That's how the plot panel implements editable titles.

jtbandes, 2022-09-08:
Related: #4174