foxglove/studio #3811: 3D beta panel starting with all subscribed

bertaveira, 2022-07-11:
**Description**
Currently the new 3D beat panel starts by subscribing to all topics. In our case we have very dense lidars that would require incredibly fast connection so when connecting to the robot wirelessly the fact that it subscribes to everything make both foxglove and rosbridge crash since it is trying to send at a data rate over 100x times the one available.

Would be nice to be able to use this panel but it would require it to start as not subscribing to anything because in live connection there is not way to even add it and unsubscribe from topics before it crashes

defunctzombie, 2022-07-11:
Related: https://github.com/foxglove/studio/issues/3692

jtbandes, 2022-09-08:
Duplicate of #3692, fixed in #3839

closed 2022-09-08 by jtbandes