foxglove/studio #3846: Broken indentation in raw messages

amacneil, 2022-07-15:
**Description**
It appears that when displaying an array with zero items, we lose the correct indentation:

[image]

- Version: main
- OS: web / macos

amacneil, 2022-07-20:
Here is a node playground script you can use to generate a broken marker with the sample dataset:

```ts
import { Input, Message } from "./types";
import { buildRosMarker } from "./markers";

export const inputs = ["/imu"];
export const output = "/broken";

export default function node(
  event: Input<"/imu">,
  globalVars: unknown
): Message<"visualization_msgs/Marker"> {
  return buildRosMarker({});
}
```

2metres, 2022-07-22:
Closed by #3841

closed 2022-07-22 by 2metres