foxglove/studio #3851: serializeStyles in emotion-serialize.browser.esm.js is called many times per frame

jhurliman, 2022-07-15:
**Description**

This method consistently shows up in profiling reports, and setting a breakpoint during playback (`3D (Beta)` is the only panel in the layout) shows it is called many times per frame. Is there something we should do differently to reduce the number of times this method needs to be called?

2metres, 2022-07-19:
Ok deep dive with @defunctzombie  on this one today. I looks as if @emotion-js (being used by @mui) rerenders very easily when the parent state updates. 

See example sandbox https://codesandbox.io/s/customstyles-demo-material-ui-tiyk49?file=/demo.tsx

Eventhough the props of the component are not changing the checkbox is re-rendering it's style function. 

Not sure what we can do right now. Long term answer is to ditch the use of @mui/styled and use am @emotion-js port of makeStyles and withStyles similar to @MUI v4's API which doesn't cause nearly as many re-renders.

Example using tss-react's makeStyles and withStyles functions which do not re-render when the parent state updates: 
https://codesandbox.io/s/customstyles-demo-material-ui-forked-fxlylz?file=/demo.tsx

defunctzombie, 2022-07-29:
@2metres can we consider this done given the merged PR #3868 or is there additional items you want to tackle in this specific bug report?

defunctzombie, 2022-08-15:
https://github.com/foxglove/studio/pull/4119

We need to remove instances of emotion styled from mui. We are aware of this now and have added some lint rules to warn us.

closed 2022-08-15 by defunctzombie