foxglove/studio #3867: [Old3d] Studio will not display single pointcloud message from MCAP

james-rms, 2022-07-19:
**Description**

The [python pointcloud example](https://mcap.dev/docs/python/jsonschema_tutorial.html) asks the user to write 10 successive messages in order to get the pointcloud to display in the old 3d panel, because if you just write one message, it doesn't display.

- Version: 1.17.1
- OS: Mac OS 12.4
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): mcap file, attached

**Steps To Reproduce**
Open a 3D panel, drag the attached MCAP onto Foxglove Studio.

[x.mcap.zip](https://github.com/foxglove/studio/files/9137141/x.mcap.zip)

**Expected Behavior**
You would see a single frame of a pointcloud.

**Actual Behavior**
No pointcloud is displayed.

james-rms, 2022-07-19:
@defunctzombie This occurs with both experimental and non-experimental players. Note that the single message does show up in the raw message panel, which indicates to me that this is specific to the 3D panel.

amacneil, 2022-07-19:
@defunctzombie for triage

defunctzombie, 2022-07-20:
I don't know that we should dedicate time to fixing up old3d if this is working in 3d beta. @amacneil my recommendation is to close this - thoughts?

amacneil, 2022-07-20:
Closing as "fixed in new 3D panel"

closed 2022-07-20 by defunctzombie