foxglove/studio #3878: 3D(Beta) render and import MESH_RESOURCE problem

LeZhang13, 2022-07-20:
**Description**

- Version: V1.18.0
- OS: Ubantu 20.04 & WIN10
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag file & ROS 1 native
- ROS distro (if applicable): Noetic & Melodic

**Steps To Reproduce**

For bag file: 1. open bag which include MESH_RESOURCE , play the bag. 2. operate the time bar.

https://user-images.githubusercontent.com/89899260/179879413-ae68b707-77af-470f-8e60-348ebad80aab.mp4

For ROS 1 Native: 1. active the node usb_cam; 2. run object detection algorithm; 3. use visulization msgs/MarkerArray MESH_RESUORCH to publish detection result (sedan、SUV、truck、people...) 4. In 3D panel is no problem , but in 3D(Beta) , it show the unexpected model in people result side.
[image]

**Expected Behavior**
Bag: Same result with 3D and 3D(Beta)
ROS 1 Native: Same result with 3D and 3D(Beta)
**Actual Behavior**
Bag: 3D(Beta) have problem
ROS 1 Native: 3D(Beta) have problem

LeZhang13, 2022-08-02:
@jhurliman Is there any idea to fix this ~

jhurliman, 2022-08-03:
This issue has not been looked at yet, but it's on the short list of bug reports to triage. Myself or whoever looks into this first will report back here with findings.

defunctzombie, 2022-08-03:
@LeZhang13 Can you please provide a bag file that shows the problem? In your description you indicate a step to open a file which includes a MESH_RESOURCE but I see no file provided. The placement of markers depends heavily on timestamps and frames so its best if we are given a file that we can use to reproduce the issue.

jhurliman, 2022-08-05:
From the information provided, this looks like a data problem and not a rendering issue. I'll convert this to a discussion and we can dive deeper into any data you can provide to help troubleshoot.