foxglove/studio #3888: 3D Beta: Merge General/Scene/Camera under Scene

amacneil, 2022-07-20:
**Description**

Clean up the 3D Beta sidebar:

- Move Frame setting (now General) into Scene
- Rename Camera to Viewport
- Nest Viewport under Scene

[image]

closed 2022-07-21 by jhurliman

amacneil, 2022-07-21:
@jhurliman what do you think about moving `Frame` under `Scene`?

jhurliman, 2022-07-22:
@amacneil I started the PR with that and @defunctzombie suggested changing it. I agreed; it was an overall degradation of the UX because it put an important field (the Render Frame) in line with other fields that are less important and shouldn't be changed often, and it required the full Scene node to be expanded which pushed more transform and topics content below the fold.

amacneil, 2022-07-22:
> @ github. com>

Ok - makes sense