foxglove/studio #3891: No image and no error is displayed if camera info is not found

amacneil, 2022-07-20:
**Description**
When looking at a bag that contains topics:

- `/camera/camera_info`
- `/camera/image_raw/compressed`

[image]

Concerns:
- If a camera topic is enabled but we couldn't find a corresponding camera info topic, we should display a topic error
- We should more aggressively search for parent topics with a camera info, e.g. in this case we need to look up one more level

jhurliman, 2022-07-20:
> in this case we need to look up one more level

What if the topics are:

`/camera/image_raw/compressed/camera_info`
`/camera/image_raw/compressed`

Should that automatically find the right topic or display a topic error?

jhurliman, 2022-07-21:
I'm going to skip the part about changing how CameraInfo topics are searched for since just implementing the topic error addresses this issue. If we come up with a new algorithm for associated topic matching we can open a new issue targeting both New3D and Image panel, since it will have implications for both.

closed 2022-07-21 by jhurliman

amacneil, 2022-07-21:
Yeah we discussed in product review today. I think we could add more intelligent topic matching, including the example you provided. I will open a separate ticket to discuss.

The new topic error looks good