foxglove/studio #3900: Improve UX around ROS 2 db3 files

amacneil, 2022-07-21:
**Description**
When loading a ROS 2 db3 file, I am presented with many "problems":
[image]

Clicking a problem tells me to use rosbridge instead:
[image]

Some thoughts on how we can improve this:
- Improve the error message explaining that custom messages are not supported with db3 files, and the options available (convert to mcap, or use rosbridge).
- Link to a longer blog post explaining the problem and solutions, including how to use [rosbag2_storage_mcap](https://github.com/ros-tooling/rosbag2_storage_mcap) directly instead of db3.

I also think we should consider dropping support for db3 entirely. I don't think Foxglove Studio is very useful without support for custom messages, and it would be simpler to direct people towards a fully supported alternative rater than providing a poor experience. Open to feedback on this.

james-rms, 2022-07-21:
Looking at this screenshot, it's complaining about [rcl_interfaces](https://github.com/ros2/rcl_interfaces) messages. It's not like these are particularly obscure custom message types - they're supported directly by the ROS2 project. One viable alternative here is to vendor those definitions into Studio, effectively punting on the problem but maintaining a decent user experience for first-time ROS2 users.

james-rms, 2022-07-21:
> I also think we should consider dropping support for db3 entirely. I don't think Foxglove Studio is very useful without support for custom messages, and it would be simpler to direct people towards a fully supported alternative rater than providing a poor experience.

One idea here is to include a copy of the mcap CLI tool in the bundle of a Studio Desktop install, such that anyone using Studio Desktop has a quick avenue to `mcap convert` any `.db3` files they have around.

amacneil, 2022-07-21:
> Looking at this screenshot, it's complaining about [rcl_interfaces](https://github.com/ros2/rcl_interfaces) messages. It's not like these are particularly obscure custom message types - they're [...]

We do actually already bundle all of [ros2/common_interfaces](https://github.com/ros2/common_interfaces). I will create a ticket to add [ros2/rcl_interfaces](https://github.com/ros2/rcl_interfaces) to the bundle.

However, this ticket is still relevant for custom messages.

> One idea here is to include a copy of the mcap CLI tool in the bundle of a Studio Desktop install, such that anyone using Studio Desktop has a quick avenue to mcap convert any .db3 files they have around.

Yeah I think that would be a good idea. We should also bundle the `foxglove` CLI.

defunctzombie, 2022-07-21:
> I also think we should consider dropping support for db3 entirely.

If you want to make that call we can do that. We pushed for this to bring along ROS2 users. It was prior to mcap so maybe we have a better story now around how a ROS2 users should interact with studio. Our analytics show that ros2-local-file users are about on par with mcap-local-file users.

defunctzombie, 2022-08-16:
I implemented some wording improvements in https://github.com/foxglove/studio/pull/4003 which I think address all the proposed suggestions.

closed 2022-08-16 by defunctzombie